stnolting/neorv32 #1157: TWD goint into idle depends on TX Fifo content

LukasP46, 2025-01-15:
First of all, I am still working out the details and may find a solution along the way.

Situation: I read (one or more) bytes from the Neorv32 as TWD.

It depends on the last read value if the twd engine.state returns to idle. It seems that the MSB of the value has to be '1' to succeed, but I haven't tested all possibilities.

The corresponding C code is really simple:
  neorv32_twd_setup(TWD_DEVICE_ID, 0, false, false, false);
  neorv32_twd_put(0x06);
  neorv32_twd_put(0x81); // won't work if < 0x80

And I just read the two bytes from a host.

I already see that the stop signal isn't noticed anymore. Like I said, I'll look into it myself, but now it's noted here for everyone to read.

stnolting, 2025-01-15:
Are you sending an ACK from the host to the TWD after reading a byte?

[image]

You example code seems to work here 🤔

LukasP46, 2025-01-16:
Ah now I see where the problem might come from. When the master does not want to read more bytes, it does not acknowledge the last transfer (for example written here in chapter 3.2.2: [mikrocontroller.net/attachment/372171/I2C_bus.pdf](https://www.mikrocontroller.net/attachment/372171/I2C_bus.pdf)).

closed 2025-01-16 by stnolting