stnolting/neorv32 #1185: RISCV_PREFIX is inconsistent with the stnolting/neorv32/sim container

Unike267, 2025-02-17:
Hi Stephan! 😃

I noticed a small issue related to the `common.mk` file.

It seems that you changed the `RISCV_PREFIX` from `riscv32-unknown-elf-` to `riscv-none-elf-`. (3703f581f6daae933ad0d87006b9c767920fa9ce)

This change directly affects compilation when using the simulation container:

`ghcr.io/stnolting/neorv32/sim:latest`

As you can see in the following image, the prefix inside the container is still using `riscv32-unknown-` instead of `risc-none`:

[image]

As a result, compilation through the current `common.mk` file using this container is broken, as shown in the following image:

[image]

If you modify the PREFIX in `common.mk`, for instance using `sed`, the compilation works fine.

[image]

### Solution

I see two possible solutions to resolve this issue:

- Revert the `common.mk` file, changing `riscv-none-elf-` back to `riscv32-unknown-elf-`.
- Modify the `PREFIX` variable from `riscv32-unknown-elf ` to `riscv-none-elf` in:

https://github.com/stnolting/riscv-gcc-prebuilt/blob/62a625936938c78fe166ca63ab1f46827b95ab59/test.sh#L7

Let me know which solution you prefer, and I’ll make a PR.

Cheers! 😄

Unike267, 2025-02-17:
Sorry for not using the bug template. I originally wrote the issue in LibreOffice, and when I went to paste it, I realized there was a template for reporting bugs (I didn’t remember that, lol).

BTW, I'm cooking a journal so I will be around for the next few months 😃 🚀

Unike267, 2025-02-17:
> https://github.com/stnolting/riscv-gcc-prebuilt/blob/62a625936938c78fe166ca63ab1f46827b95ab59/test.sh#L7

Sorry my bad, this is not the file that It should be fixed. I thought that this was the script that generetes the compiled gcc. Now I'll take a closer look.

Unike267, 2025-02-17:
Sorry, Stephan, I have a thousand things on my mind. I found a script that sets a PREFIX and used it without looking at it carefully. 😥

I just realized now, reviewing the issue on my way home on the train, that it's a test script, LOL.

By the way, I can't find the Makefile (or similar) that compiles GCC for RISC-V, where we should change the PREFIX.
I see that the resulting `.tar` file is uploaded to the [release](https://github.com/stnolting/riscv-gcc-prebuilt/releases).
I looked for the CI file that uploads this asset to deduce the script that generates the `.tar` and point you to the line that needs to be changed. You don't upload the asset manually, do you?

Sorry again for the confusion!

stnolting, 2025-02-20:
No worries! It happens to me all the time too...
This project is sometimes just too big and overwhelming/chaotic. 😅 

> By the way, I can't find the Makefile (or similar) that compiles GCC for RISC-V, where we should change the PREFIX. I see that the resulting .tar file is uploaded to the [...]

I built the toolchains from this repository myself and then uploaded them. But since this is too time-consuming for me and I don't want to bother with the gcc build system every time, I already switched to [xPack](https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack) for the main NEORV32 repo. This is also the plan for all other "sub-repositories". Once xPack is used everywhere, this repository can be deleted/archived. So consider it deprectaed. :wink:

Unike267, 2025-02-21:
> I already switched to [xPack](https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack)

Oh, that’s so interesting! I didn’t know about this tool, I’ll have to switch too 😄.

I wonder if there’s a container for it.

If not, we could modify `sim.dockerfile` to something like this:

```diff
FROM ghcr.io/hdl/debian/bullseye/sim/osvb

RUN apt-get update -qq \
 && DEBIAN_FRONTEND=noninteractive apt-get -y install --no-install-recommends \
  g++ \
  git \
  make \
  python3-pip \
  time \
 && apt-get autoclean && apt-get clean && apt-get -y autoremove \
 && rm -rf /var/lib/apt/lists/* \
 && pip3 install wheel setuptools \
 && pip3 install doit \
 && mkdir -p /opt/riscv \
- && curl -fsSL https://github.com/stnolting/riscv-gcc-prebuilt/releases/download/rv32i-131023/riscv32-unknown-elf.gcc-13.2.0.tar.gz | \
+ && curl -fsSL https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/download/v14.2.0-3/xpack-riscv-none-elf-gcc-14.2.0-3-linux-x64.tar.gz | \
 tar -xzf - -C /opt/riscv \
 && ls -al /opt/riscv

ENV PATH $PATH:/opt/riscv/bin
```

I’ll take a closer look. Maybe @umarcor can give some hint/tips, then I’ll ask him at the lab.

Unike267, 2025-02-21:
I've already built a container with this containerfile and works fine!

[image]

If you agree, I'll do a PR!

Unike267, 2025-02-21:
(Ignore the reference to the `f382926` commit. Before making the PR, I noticed that parentheses in the links of the description were included in the link when opened through `gitk`. So, I modified the description, force-pushed without parentheses in the links, and then submitted the PR.)

closed 2025-02-22 by stnolting