stnolting/neorv32 #1195: Bootloader does not support SPI Flash WP Pin

jpf91, 2025-03-04:
**Is your feature request related to a problem? Please describe.**
When playing with the NEORV32 on the Tang Nano 20k board again, I noticed that the bootloader can not save the application to flash.

The (low-active) WP Pin on the board does have a pull-up resistor, but it still seems to be driven low when the FPGA pin is not configured. Unfortunately I don't have anything here to measure the signal, but writing and then loading the flash magically starts to work if I explicitly drive the WP pin from the boatloader.

**Describe the solution you'd like**

I've added code to the booatloader to explicitly drive the WP pin: https://github.com/jpf91/neorv32/commit/12a2f3145b7e122596b2b505eb0af92fa54fe9d8

This seems to work, but it pushes the bootloader over the 4K ROM limit..... Any ideas what to do?

**Describe alternatives you've considered**

I tried just hardcoding the pin output to '1' in the VHDL top-level file. Unfortunately this seems to have caused corruption the FPGA bitstream. I'm not sure why, but it seems to interfere with the programmer.

**Additional context**

A complete repo with neorv32 for the Tang Nano 20k can be found here: https://github.com/jpf91/neorv32-tang20k
There's no proper documentation yet, but this blog post explains the setup: https://www.jpfau.org/blog/tang20k-neorv32/

stnolting, 2025-03-04:
Hey @jpf91!

> The (low-active) WP Pin [...]

I haven't even thought about this situation yet... But that seems to be a general problem with SPI memories.

> I've added code to the booatloader to explicitly drive the WP pin:

Using a GPIO output is great work-around, but it is also a very setup-specific solution I think. 🤔

We could add another “WP” signal to the SPI and TWI controllers that can be controlled by software. But somehow that feels over-engineered...

> I tried just hardcoding the pin output to '1' in the VHDL top-level file. Unfortunately this seems to have caused corruption the FPGA bitstream. I'm not sure why, but it seems to interfere with the programmer.

This was also my first thought. But as you write, this seems to interfere with the bitstream configuration. It's no different with AMD FPGAs - you need a special primtive to access all the SPI lines of the bitstream flash. Maybe there something similar in Tang Nano FPGAs?

Could you use one of the reset **outputs** of the processor for controlling WP?

https://github.com/stnolting/neorv32/blob/5753b32bd4aeac0b6d63a355fdb4038ed80b1df0/rtl/core/neorv32_top.vhd#L146-L147

These outputs should be high-Z during configuration and will become high as soon as the processor has completed its internal reset cycles. But maybe we'll end up having the same problem as with your hardwired approach.

jpf91, 2025-03-08:
Thanks for the hint! Indeed a simple `mspi_wp <= '1' when rstn_wdt_o = '1' else '0';` works fine on the Tang Nano 20K Board!

closed 2025-03-08 by jpf91

stnolting, 2025-03-08:
> Thanks for the hint! Indeed a simple mspi_wp <= '1' when rstn_wdt_o = '1' else '0'; works fine on the Tang Nano 20K Board!

Great to hear!

If you like, you can add your setup to [neorv32-setups](https://github.com/stnolting/neorv32-setups). We already have support there for the [Tang Nano 9k](https://github.com/stnolting/neorv32-setups/tree/main/gowineda/tang-nano-9k), but not for the Tang Nano 20k.

jpf91, 2025-03-09:
> If you like, you can add your setup to [neorv32-setups](https://github.com/stnolting/neorv32-setups). We already have support there for the [Tang Nano 9k](https://github.com/stnolting/neorv32-setups/tree/main/gowineda/tang-nano-9k), but not for the Tang Nano 20k.

I'll open a PR for this in a few days. I now even managed to reuse the on-board JTAG programmer that's used for the FPGA with OpenOCD and gdb to debug SW applications.
So with this board, you basically have:
* FPGA Programmer
* UART
* JTAG Debugger for SW

All on a single USB connection and with an Arduino Nano Sized board. That's quite nice :-)

stnolting, 2025-03-10:
> I'll open a PR for this in a few days.

Great! Thank you very much!

I'm also thinking about getting one of those Tang Nano boards. So this would be very helpful.

> So with this board, you basically have: [...]

How is this handled? Is there something like a four-channel FTDI chip?
Are there different interfaces/pins for bistream upload and processor debug?