stnolting/neorv32 #1199: More than two cores

pepijndevos, 2025-03-05:
I'm quite interested in making a SoC with more than two cores.

Looking at the code a bit it seems like maybe all that would be needed is to generalize the bus switch for more than two cores?

Maybe inter-core communication would require some thought as well.

stnolting, 2025-03-05:
Hey @pepijndevos!

> I'm quite interested in making a SoC with more than two cores.

The simplest approach would be to instantiate several NEORV32 tops, connect them to a global memory and daisy-chain the JTAG chain. This would then no longer be an SMP architecture, but rather a NUMA architecture.

> Looking at the code a bit it seems like maybe all that would be needed is to generalize the bus switch for more than two cores?

Basically, you are right. The on-chip-debugger's debug module already supports up to 4 CPU cores (but that could be easily extended):

https://github.com/stnolting/neorv32/blob/8ab1fecf6832164c4c97c8c61c67afd60369ef2d/rtl/core/neorv32_debug_dm.vhd#L21-L23

The core-local interrupt supports even more CPU cores:

https://github.com/stnolting/neorv32/blob/8ab1fecf6832164c4c97c8c61c67afd60369ef2d/rtl/core/neorv32_clint.vhd#L21-L23

> Maybe inter-core communication would require some thought as well.

Right. That would require a more complex fully-connected archiitecture since every core needs to be able to communicate with every other core.

----------

When we introduced dual-core support, I did a few experiments with even more cores.

Two cores (without any CPU caches) surprisingly rarely get in each other's way (in terms of bus congestion). I think this is due to the fact that the CPU is a multi-cycle architecture that does not perform bus accesses in each and every cycle. With four cores, however, there was already a significant congestion in the central bus system.

With bigger/better CPU caches, however, you could certainly compensate for some of this.

stnolting, 2025-09-06:
I think this can be closed.

closed 2025-09-06 by stnolting