stnolting/neorv32 #1203: Boot Modes Not Working with Vivado IP Core, but Works with HDL Implementation

Muddassar180009, 2025-03-11:
Problem Summary:
I am working with the NEORV32 RISC-V Processor, and I am facing an issue when trying to use IMEM Boot Mode with the Vivado IP core. The HDL-based implementation works fine for both Booting via Bootloader and IMEM Boot Mode, but when using the Vivado IP core, IMEM Boot Mode does not work as expected. Additionally, while the Booting via Bootloader mode runs into issues with the neorv32_rte_setup() function, it works fine by commenting this function.

Behavior in Different Setups:
HDL-Based Implementation:
When using HDL code and running in Boot IMEM Image Mode or Booting via Bootloader Mode, everything works as expected.
No issues are encountered with the neorv32_rte_setup() function being used.

Vivado IP Core:
IMEM Boot Mode does not work when using the Vivado IP core. The program does not execute when the IMEM Boot Mode is selected in IP Core GUI.
The Booting via Bootloader mode shows an issue when neorv32_rte_setup() is called. Although I can upload the .bin file, I get an RTE-ERR and the program doesn’t run as expected.
If I comment out neorv32_rte_setup(), the Booting via Bootloader mode works fine, but the IMEM Boot Mode still doesn’t work with the Vivado IP core.

Setup Details:
Vivado Version: Vivado 19.1
Processor Mode: NEORV32 RISC-V Processor
Program Used: sw/example_demo_blink_led
Hardware Setup: Simple setup using GPIOs and UART0 with a .bin file of the demo program.

Can someone help explain why IMEM Boot Mode is not working when using the Vivado IP core, but works fine with the HDL-based implementation? I would appreciate any advice on how to properly configure the Vivado IP core setup to make IMEM Boot Mode work and resolve the RTE-ERR in Booting via Bootloader mode.

Thank you for your help!

[image]

stnolting, 2025-03-11:
Another (more flexible) option is to use an external memory for booting.

## Boot mode: Custom Address

1. Select the "Custom Address" boot mode in the NEORV32's general tab:

[image]

2. Here we use the default boot address `0x00000000`. Thus, we do not need to change any linker options. However, this means we need to **disable** the internal IMEM as we want to boot from an external memory:

[image]

3. Connect a block memory to the processor's AXI system. For example:

[image]

4. Compile your applition firmware and generate a COE memory initialization file:

```bash
neorv32/sw/example/hello_world$ make clean coe
Memory utilization:
   text    data     bss     dec     hex filename
   5700       0     116    5816    16b8 main.elf
Generating neorv32_raw_exe.coe
```

5. Double click on the memory block module in the block design and select "Single Port ROM" as memory type:

[image]

6. Go to the "Other Options" tab, enable "Load Init File" and select the COE file you've just created:

[image]

7. Click OK to close the customization window. Go the design's address editor. Ensure that your memory module is correctly mapped to the configured boot address (`0x00000000` in this case):

[image]

stnolting, 2025-03-11:
I hope this helps.

Muddassar180009, 2025-03-12:
Hi @stnolting ,

Thank you so much for your quick and detailed response! I really appreciate the time and effort you took to explain the steps thoroughly. I followed both methods you mentioned, but unfortunately, neither has worked for me.

Regarding the first method you mentioned (IMEM Boot Mode), I had already tried that approach as described in my original post. I followed the same steps, but the IMEM Boot Mode is still not functioning as expected. I did recheck everything, including running the necessary scripts and refreshing the IP catalog, but the issue persists.

As for the second method, where I boot from a custom address, I followed all the steps, disabled the internal IMEM, and ensured the memory block is correctly mapped to the boot address (0x00000000). Unfortunately, this approach also did not work, and the system does not boot as expected.

I have attached the pictures for reference. Could you kindly guide me on where I might be making a mistake? I would greatly appreciate any additional insights or suggestions you could offer.

I apologize if I'm making any immature mistake in this process, as I am fairly new to this field, and I’m eager to learn and improve. Thank you again for your assistance, and I look forward to your guidance on the next steps.

[image]

[image]

[image]

[image]

[image]

stnolting, 2025-03-12:
Your setups loosk good so far.

Could you also share a screenshot of the "CPU" tab so we can check the ISA configuration?

Are there any (critical) warnings when running synthesis/implementation?
Did the design meet timing constraints?

Can you do a simulation of your setup?
Alternatively, you could attach an ILA to the processor's AXI port to monitor the bus transactions from the hardware manager.

Unike267, 2025-03-12:
According to `UG: 12. Packaging the Processor as a Vivado IP Block`, the set frequency is 100 MHz.

Could you perform a test at this frequency instead of 200 MHz?

Just to rule out timing issues.

Muddassar180009, 2025-03-12:
> Your setups loosk good so far.
> 
> Could you also share a screenshot of the "CPU" tab so we can check the ISA configuration?
> 
> Which version of the processor are you using?
> 
> Do you have a more recent Version of Vivado you could test?
> 
> Are there any (critical) warnings when running synthesis/implementation? Did the design meet timing constraints?
> 
> Can you do a simulation of your setup? Alternatively, you could attach an ILA to the processor's AXI port to monitor the bus transactions from the hardware manager.

I am using v1.11.1 of this repo. 
The picture of CPU tab is attached. At this moment i have 2019.1 of vivado.
Yes there is only one critical warning of timing violation, **([Timing 38-282] The design failed to meet the timing requirements. Please see the timing summary report for details on the timing violations.** But this warning is removed when i change frequency to 100MHz.
I was doing behavioural simulations and same behaviour appear there as well **nothing on uart as well as gpio_out = 0;**

[image]

Muddassar180009, 2025-03-12:
> According to `UG: 12. Packaging the Processor as a Vivado IP Block`, the set frequency is 100 MHz.
> 
> Could you perform a test at this frequency instead of 200 MHz?
> 
> Just to rule out timing issues.

I tried at lower frequency (100 MHz) as well, but behaviour remains same :(

stnolting, 2025-03-12:
Please **disable** the `E` ISA extension:

[image]

The CPU will crash when executing code, that was **not** compiled with the `E` extension (which is the default).

If this also does not fix the problem, then please enable the bootloader-based boot configuration and check if you can anything happening on the UART.

Muddassar180009, 2025-03-12:
> Please **disable** the `E` ISA extension:
> 
> [image]
> 
> The CPU will crash when executing code, that was **not** compiled with the `E` extension (which is the default).
> 
> If this also does not fix the problem, then please enable the bootloader-based boot configuration and check if you can see any bootloader UART messages.

This works for me. By disabling the "E" ISA Extension, all three boot modes are working now.
Thanks a lot for your help..,

stnolting, 2025-03-12:
Great to hear! :+1:

So we can close this?

closed 2025-03-12 by stnolting