stnolting/neorv32 #1216: Makefile error, readlink does not exist in power shell

ecstrema, 2025-03-24:
**Describe the bug**
I have a Linux setup for compiling the neorv32's software, but my supervisor in on windows, and it failed when he tried to run it with powershell. 

It fails because the readlink command, used to check if sh is available does not exist in powershell. 

Git bash works though, so that's an easy workaround. 

I don't know what's the best way to check if a target exists in powershell, but I'll have a look. Symlinks seem like a concept almost foreign to Windows. 

**To Reproduce**
Run make check in powershell. 

**Expected behavior**
Works if you have gcc and make installed and in your path. 

**Environment:**
 - OS: Windows 11, powershell

ecstrema, 2025-03-24:
Well indeed, I wonder too. I'll try to see if anything fails of if I remove this check tomorrow. 

If it doesn't work, I think it's still better to advise people to use git bash or similar instead of explicitly supporting powershell.

stnolting, 2025-03-29:
I've removed the "shell check". It looks like this was really just an artifact from the early days with no real function.

closed 2025-03-29 by stnolting