stnolting/neorv32 #1218: AHB-Lite Bus Integration in neorv32_vivado_ip

Muddassar180009, 2025-03-27:
I am trying to modify the neorv32_vivado_ip to use the AHB bus instead of the AXI4-Lite bus, as the guide suggests using AXI4-Lite. I've made the following changes to adapt the design to AHB-Lite:

In the TCL script, I modified:
Line 55: Added the AHB bridge instead of the AXI4-Lite bridge:
**add_file $neorv32_home/rtl/system_integration/xbus2ahblite_bridge.vhd**
Line 125: Updated the bus interface:
**set_property enablement_dependency {$XBUS_EN} [ipx::get_bus_interfaces ahb_lite -of_objects [ipx::current_core]]**
Line 210: Modified the group name for the bus interface:
**set group [add_group $page {External Bus Interface (XBUS / AHB-Lite-MM Host)}]**

In the neorv32_vivado_ip.vhd file, I replaced all instances of AXI4 with AHB in the module and signal names.

[image]

[image]

[image]

However, I am encountering the following error:
**ERROR: [Common 17-55] 'set_property' expects at least one object. 
Resolution: If [get_<value>] was used to populate the object, check to make sure this command returns at least one valid object.**

It seems that the set_property command is failing because the object being referenced (like the AHB interface) is not being found or properly instantiated. I’ve double-checked the changes, but I’m still unable to resolve the issue.

Could you please guide me on the correct approach to integrate AHB-Lite and address this error?

Thanks for your help!

stnolting, 2025-03-27:
This might be the problem:

```tcl
set_property enablement_dependency {$XBUS_EN} [ipx::get_bus_interfaces ahb_lite -of_objects [ipx::current_core]]
```

I think Vivado cannot identify your `ahb_*` top ports as AHB-lite interface. You need to stick to a very specific naming convention; otherwise Vivado cannot identify those interfaces.

Please change that line to a generic port (just to check that this is actually the reason for the error message):

```tcl
set_property enablement_dependency {$XBUS_EN} [ipx::get_ports ahb_* -of_objects [ipx::current_core]]
```

Muddassar180009, 2025-03-28:
Thank you so much for your previous help; it worked perfectly for generating the IP!

However, I am now encountering a new issue. After combining all the AHB pins with the AHB Lite bus, I am unable to connect my NEORV32 IP to Vivado’s AHB Lite to AXI Bridge. I keep receiving the error **"No matching connections"** when attempting to link the ports. I suspect this could be due to a mismatch with the hready_in and hsel signals.

Additionally, when I manually connect all other signals, I notice in the address editor that the BRAM is shown as an unconnected slave.

[image]

[image]

Do you think this issue is due to a port mismatch between the AHB interface of the NEORV32 and the Bridge interface, or could there be another underlying problem?

I would appreciate your insights on this matter.

stnolting, 2025-03-29:
> Thank you so much for your previous help; it worked perfectly for generating the IP!

Great to hear!

> I keep receiving the error "No matching connections" when attempting to link the ports.
> I notice in the address editor that the BRAM is shown as an unconnected slave.

This is because Vivado cannot identify the `ahb_*` ports as an actual AHB interface. You can do that manually in the IP packager or can do that by using some attributes in the VHDL code. You can find an example in the Vivado language templates:

[image]

I have nerver used AHB in a Vivado IP design so I am not sure what the correct "interface name" is so that Vivado can actually identify all the ports.

How about using the AXI-to-AHB bridge instead (at the cost of additional latency)? 😉

Muddassar180009, 2025-04-01:
Thanks for your response. Let me try some try some attributes in HDL.

stnolting, 2025-04-04:
Keep us updated!

stnolting, 2025-09-06:
The AHB interface was removed because it cannot support all features of the XBUS protocol (e.g., bursts). A complete AHB bridge would be great, but that is not planned at the moment.

Closing this (for now?).