stnolting/neorv32 #1240: Exception upon executing binary uploaded via bootloader

henrikbrixandersen, 2025-04-22:
**Describe the bug**
When uploading and executing a binary via the latest NEORV32 bootloader update, a hardware exception is thrown.

Reverting to the previous, pre-compiled NEORV32 bootloader (reverting commit f99b8f5d0af0f274225db04ababb3ccce7c2281b) and uploading the same binary works.

**To Reproduce**
- Build `neorv32/sw/example/hello_world`
- Upload via UART to NEORV32 bootloader and execute
- Observe hardware exception, see below

**Expected behavior**
Upload and jump to uploaded application succeeds.

**Screenshots**
```
NEORV32 Bootloader

BLDV: Apr  5 2025
HWV:  0x01110300
CLK:  0x0112a880
MISA: 0x40901100
XISA: 0x00000083
SOC:  0x001f800d
IMEM: 0x00010000
DMEM: 0x00010000

Autoboot in 10s. Press any key to abort.
Aborted.

Available CMDs:
 h: Help
 r: Restart
 u: Upload via UART
 s: Store to SPI flash
 l: Load from SPI flash
 e: Start executable
CMD:> 
e
Booting from 0x00000000...

ERROR_EXCEPTION 0x00000002 0x00000000 0xf14020f3 0x00000000
```

**Environment:**
 - OS: Linux
 - GCC Version (RISC-V and native): riscv-none-elf-gcc-14.2.0
 - Libraries used (clib, newlib, ...): None

**Hardware:**
 - Hardware version: v1.11.3 (2c0b0e75a91fe08ac77e12a657e891f116b2945a)
 - Implemented CPU extensions, peripherals: UP5Kdemo processor template
 - Hardware modifications: None

**Additional context**
Running on the iCEBreaker board.

stnolting, 2025-04-22:
This is interesting...

```
ERROR_EXCEPTION 0x00000002 0x00000000 0xf14020f3 0x00000000
                mcause     mepc       mtinst     mtval

mcause: 0x00000002 = illegal instruction exception
mepc:   0x00000000 = beginning of IMEM
mtinst: 0xf14020f3 = csrrs x1, mhartid, x0
```

The CPU tries to read the hart ID which causes an illegal instruction exception. This can only happen if the CPU is in user mode (since `mhartid` can only be read by machine-mode code). And that would mean that there is some bug when the bootloader (running in machine-mode) launches the application code (which should also start in machine-mode). I will have a closer look at this.

stnolting, 2025-04-22:
I can confirm the error. Apparently we didn't test properly. 🙈

closed 2025-04-22 by stnolting