stnolting/neorv32 #1242: Consider leveraging VexRISC-V CPU core?

bitandquit, 2025-04-22:
Hi Steve,

Thanks for work on NeoRV32, it really sets a high bar of quality of implementation and documentation.

At the risk of insulting your work, have you ever considered leveraging the open-source VexRISC-V core?  I ask since this CPU core seems to achieve a very high performance/area/Fmax result.

I'm still a fan of NeoRV32 and will keep using it, but there have been a few times where getting a CPI ~1 would be a benefit.

Kindly regards!

stnolting, 2025-04-22:
Don't worry, I'm not taking anything personally here. :wink:

I have never worked with VexRISC myself, so my statements should be taken with a grain of salt. A colleague once used it in a project. Compared to the same configuration of NEORV32, VexRISC was larger and ran at a lower clock rate. But as I said, I haven't worked with it myself and I don't know the details. Anyway, I think it's a pretty impressive project.

When choosing a soft-core processor, you should ask yourself what you really need. GitHub is full of different cores that serve all kinds of application areas. If you need a (Linux-capable) core with a lots of power, then use VexRISC, if you only need a programmable state machine, then use SERV. I see the NEORV32 somewhere in between in its very own niche and I would like to stick with it.

So no, I'm not planning to replace the entire CPU core. But perhaps some aspects can be adopted from VexRISC.

bitandquit, 2025-04-23:
Hi Steve,

First, thanks for fast reply and considering my request.  To give you a bit of insight, my intent isn't to run Linux (which I agree needs a much larger core) but for real-time tasks.

I've found ~10 MIPS+ is pretty ideal which means running NeoRV32 @ 50MHz+ due to a CPI ~4-5.  This is of course inline with TI's physical MSP430 (and your Neo430 too 😄 ) but AVRs and Cortex-M0s do about ~1 CPI so they can be clocked lower for the same work load.

For some "deeply embedded" systems targetting older / slower FPGAs having to run or close timing at 50 MHz+ can take effort and uses more power.

Not that I'm suggesting VexRISC is the "best" solution but the processor part seemed to be inline resource wise with NeoRV32 (~1K FF/ ~1K LUT territory) with reasonable speeds.  f32c (https://github.com/f32c/f32c) is exciting from a performance/area point of view but I keep coming back to NeoRV32 because it's a high quality "everything in one place" solution including the all important RISC-V debug capability, SVD definitions for all peripherals, linker scripts and of course your patience in answering questions! 😆 .

Just some food for thought.

stnolting, 2025-04-23:
> I've found ~10 MIPS+ is pretty ideal which means running NeoRV32 @ 50MHz+ due to a CPI ~4-5

I would love to have more performance. But not at the expense of much bigger hardware or a much lower clock frequency. I have often made concepts of a "real" pipeline (with 2 or 3 stages), but all the hazards and especially the exception handling make the logic bigger. And the bus interface would also have to be adapted, which in turn makes the memory system more complicated... There is no free meal. 😅

Anyway, I am absolutely open to suggestions on how to further improve performance - even if that means rewriting the whole core.

> Not that I'm suggesting VexRISC is the "best" solution but the processor part seemed to be inline resource wise with NeoRV32 (~1K FF/ ~1K LUT territory) with reasonable speeds. f32c (https://github.com/f32c/f32c) is exciting from a performance/area point of view

I agree!

But I always find it difficult to draw comparisons. Which RISC-V exceptions are supported? Does the core check (all) illegal instructions? What happens if bus accesses return an error? How were the CPI/IPC values calculated? With an "ideal" bus system (no wait states and multi-layer interconnect that allows parallel accesses)? And so on and so forth.

In the end, you usually have no choice but to test and benchmark the system (on an FPGA) with the actual application...

closed 2025-06-19 by stnolting