stnolting/neorv32 #1258: bootloader jumps to SPI flash loading mode

mahdi259, 2025-05-04:
**Describe the bug**
I packaged neorv32 into an IP following user guide documentation and implemented the following block design.

I have an issue with processor boot. It doesn't count 10 seconds and directly jumps to SPI flash loading mode. 

**Expected behavior**
It should wait for 10 seconds. 

**Screenshots**
The output:
[image]

System block design:
[image]

Memory mapping:
[image]

CPU configurations:
[image]

[image]

[image]

Thanks

mahdi259, 2025-05-04:
I checked v1.10 version and in that version this problem is not present.

stnolting, 2025-05-05:
Let's stick to version 1.11.3.8 here.

Your Vivado IP configuration looks fine so far.

> `ERROR_EXCEPTION 0x00000005 0xffe00684 0x00072783 0xfff80000`

This is a "load access fault" exception that occured while accessing address `0xfff80000`, which is the base address of the SPI module.

According to the bootloader splash screen the SPI module is not implemented:

`SOC:  0x00028003`

Bit 18 is zero indicating that the SPI module is not there.

https://github.com/stnolting/neorv32/blob/e13613a595a8bc4bb8db5bcf7ed67d22beebea36/sw/lib/include/neorv32_sysinfo.h#L62

So you need to enabled SPI in the "peripherals" tab of the IP configuration dialog.

stnolting, 2025-05-05:
> It doesn't count 10 seconds and directly jumps to SPI flash loading mode.

That is because the CLINT is not enabled either (bit 16 in the bootloader's SOC line is zero):

https://github.com/stnolting/neorv32/blob/e13613a595a8bc4bb8db5bcf7ed67d22beebea36/sw/lib/include/neorv32_sysinfo.h#L60

stnolting, 2025-05-05:
[image]

This is misleading and should not be printed if the CLINT is not available. I'll fix that message.

mahdi259, 2025-05-05:
Thanks a lot. I enabled CLINT and now it counts 10 seconds. BTW, now it doesn't jump to SPI flash loading mode! (SPI flash is not enabled)

closed 2025-05-05 by mahdi259

mahdi259, 2025-05-05:
I'm really thankful for your great support. Best wishes for you.