stnolting/neorv32 #1261: A question about RAM base address modification

mahdi259, 2025-05-09:
Hi
To change the RAM base address from 0x80000000 to 0x10000000, are inneorv32_package.vhd and neorv32.ld the only files one should change?
Thanks a lot

stnolting, 2025-05-10:
Are you using a Vivado IP setup with MIG?

Actually, you don't have to change the linker script nor the VHDL package file. Just disable internal memories (at least the DMEM).

If you are using an AXI-based setup you need to define an address map for the interconnect. If your RAM is 2GB you need to map it to a base address that is aligned to 2GB boundary. Hence, you can only use `0x0000000` and `0x80000000` (note that when using `0x80000000` the last kBs of that address space will overlap with the processor-internal peripherals).

The processor does not care about the RAM mapping. You just need to compile your code accordingly. You can define the mapping and layout in the firmware's Makefile. For example, you could use the lower half of your DDR RAM as "virtual IMEM" and the upper half as "virtual DMEM":

```makefile
# Adjust processor IMEM size and base address
USER_FLAGS += -Wl,--defsym,__neorv32_rom_size=1G
USER_FLAGS += -Wl,--defsym,__neorv32_rom_base=0x00000000

# Adjust processor DMEM size and base address
USER_FLAGS += -Wl,--defsym,__neorv32_ram_size=1G
USER_FLAGS += -Wl,--defsym,__neorv32_ram_base=0x40000000
```

mahdi259, 2025-05-10:
> Are you using a Vivado IP setup with MIG?
Yes.
You're right. Thanks a lot.

closed 2025-05-10 by mahdi259

mahdi259, 2025-05-10:
> The processor does not care about the RAM mapping. You just need to compile your code accordingly. You can define the mapping and layout in the firmware's Makefile. For example, you could use the lower half of your DDR RAM as "virtual IMEM" and the upper half as "virtual DMEM"

The only problem with this method is that ICACHE will get higher miss rate since IMEM and DMEM are located inside the ICACHE SPACE. right?

stnolting, 2025-05-10:
Internal IMEM and DMEM are also cached by the CPU caches. But if IMEM & DMEM are not enabled, the address space that they would actually occupy is also mapped to the external interface.

Have a look at this image (from the data sheet). When IMEM and DMEM are disabled, the blue boxes are just gone and their address space also becomes part of the "void", which in turn is available for the external bus interface.

[image]

mahdi259, 2025-05-10:
My understanding is that DCACHE looks at mem_dmem_base_c (0x80000000 by default) and caches the addresses starting from it. 

So if I virtually compile the program to use 0x40000000 as base address of data memory, then DCACHE won't cache those addresses (since the neorv32_package.vhd is unchanged) and ICACHE will cache both instructions and data. 

```
# Adjust processor IMEM size and base address
USER_FLAGS += -Wl,--defsym,__neorv32_rom_size=1G
USER_FLAGS += -Wl,--defsym,__neorv32_rom_base=0x00000000

# Adjust processor DMEM size and base address
USER_FLAGS += -Wl,--defsym,__neorv32_ram_size=1G
USER_FLAGS += -Wl,--defsym,__neorv32_ram_base=0x40000000
```

This way DCACHE is not used and ICACHE will experience high miss rates.

stnolting, 2025-05-10:
> My understanding is that DCACHE looks at mem_dmem_base_c (0x80000000 by default) and caches the addresses starting from it.

The caches _cache_ **the entire address space** (or to be more precise: that part of the address space that is marked "cached" in the figure above; so `0x00000000` to `0xEFFFFFFF`).

If your program performs a load targeting any address below `0xF0000000` - it will be cached.
If your program executes an instruction from any address below `0xF0000000` - it will be cached.
There are exceptions (atomic memory accesses) but they don't matter here.

mahdi259, 2025-05-10:
Sorry, how does the processor know to which cache (ICACHE or DCACHE) should assign an specific content that's loaded from xbus?

stnolting, 2025-05-10:
Actually, that's a good question!

The CPU core has two interface - one for fetching instructions and another one for loading/storing data. The instruction interface is directly connected to the i-cache and the data interface is directly connected to the d-cache.

[image]

If the CPU requests a new instruction from some address, this request is sent to the i-cache. Then, the i-cache will fetch the according memory block from main memory (if it lies in the "cached" address space).

If the CPU want to load/store some data, this request is sent to the d-cache. Then, the d-cache will fetch the according memory block from main memory (if it lies in the "cached" address space) when it is a load operation. In case of a store, the d-dache will forward that request directly to main memory ("write-through").

mahdi259, 2025-05-10:
Thanks a lot for your clarification.

stnolting, 2025-05-10:
You're welcome!

By the way, I think it's pretty stupid that we have the concept of IMEM and DMEM. 🙈 The CPU doesn't care where data and instructions are located. IMEM and DMEM is just memory at some addres.

But that's still a relic from the early days of this project...

mahdi259, 2025-05-10:
The way CPU handles instruction/data is nice 👍. Have a good night.