stnolting/neorv32 #1269: Instruction fetch from xbus external memory during execution falls into bad 8 bit accesses.

kurtjensen, 2025-05-16:
First let me thank snolting and the community for providing such a well documents soft core.  This is excellent work and I really do appreciate what you all have done here!

I am running an Artix 7 on the QMTech board with the XC7A15T on it, clocked at 50 MHz, no PLLs.  I have added an external SRAM (ISSI IS62WV51216BLL-55TLI) over Wishbone, and wrote some glue HDL to place it at 0x90000000 - 0x900FFFFF in the address space.  It supports 8, 16 and 32 bit aligned accesses, and holds the wishbone ACK for a few clocks to be able to do a second transaction to this 16 bit-wide memory. 

Neorv32 is built in vivado with single core, UART and SPI, and debug engine generics on, memory cache off, clock freq set to 50MHz. ISA has z,m,u and Zicntr chosen. HDL builds well with over 9ns of slack reported once fit into the FPGA.

I added a command to the bus_explorer example code (that executes out of on-FPGA memory) to test access to this external RAM. Using this, I have convinced myself that it can read and write at all of these data widths without error.

The problem comes when writing another c application that is linked so it is located in this external memory instead of in on-FPGA RAM.  Execution into crt0.S and then on into main() all seems to work, but a few lines into main() I encounter mcause=2 type traps.  

main() is not trying to do anything remarkable here, just simply reproduce a typical simple "Hello World" executable using neorv32_rte_setup(), neorv32_uart_setup(NEORV32_UART0, BAUD_RATE, 0), and neorv32_uart_printf().  I'm using the xpack stuff (riscv-none-elf-gcc v 14.2.0 and newlib 4.4.0).

After much gnashing of teeth playing with the code, gdb use, new linker scripts, and crt0.S all giving non-reproducible and contradictory results, I finally resigned myself to adding a debug core in Vivado and watching the memory transactions when it was failing.

[image]

I apologize for the eye chart screen capture of the vivado logic analyzer output.  Code lives at 0x90000000, and stack at the other end down from 0x900FFFFF.  You can see that an instruction fetch starts at clock cycle 1553 in the capture (where the yellow cursor is) with wb_stb rising. When wb_stb falls on the next clock cycle, we see that the xbus logic is sending wb_sel[3:0] = 8, which becomes a byte-wide instruction fetch. I expect this should be a 32 bit wide fetch (or perhaps it might be 16 bit fetch for a compressed instruction ? - I am still learning RISC-V :) ) The previous cycle is a byte-wide write to the stack, which I have not figured out yet.

Please note also the subsequent jump to the trap handler at 0x900009a0 and the tight loop there that follows.  Here, starting around clock cycle 1590, we see wb_sel[3:0] = 8 where I also would have expected 16 or 32 bit instruction fetches. 

From what I can tell, this means simply that the whole instruction bit pattern is simply not being fetched from memory when this happens.

I suspect this is a bug in driving the wb_sel lines correctly during cpu execution from external memory.  

I want to let the community know about this, so I am posting here. I could not find other evidence on the web of others using this soft core and executing out of external memory, so perhaps this is a bit bleeding edge for this project? Alas, I may not be able to devote much time in helping debug this issue, but I will try to help some as my time an resources allow.

Regards,
-Kurt

kurtjensen, 2025-05-16:
I almost forgot:  My git clone of neorv32 was from fc5bc029e56849e6e00e4c3bc5e181a0065acff0 

Regards,
_K

stnolting, 2025-05-17:
Hey @kurtjensen,

thank you very much for the nice words!

> Neorv32 is built in vivado with single core

So you are not using the defualt Vivado IP module (with AXI interface), right?

> I added a command to the bus_explorer example code (that executes out of on-FPGA memory) to test access to this external RAM. Using this, I have convinced myself that it can read and write at all of these data widths without error.

The bus explorer supports byte, half-word and word accesses. Did you check all three of them with your SRAM controller?

> The problem comes when writing another c application that is linked so it is located in this external memory instead of in on-FPGA RAM. Execution into crt0.S and then on into main() all seems to work, but a few lines into main() I encounter mcause=2 type traps.

Did you map the entire firmware to the SRAM (code and data)? Does the same problem occur if you execute code from IMEM but use your SRAM for data?

> I finally resigned myself to adding a debug core in Vivado and watching the memory transactions when it was failing.

I feel you. ILAs are just great for debugging those issues! 😅

> [...] screen capture of the vivado logic analyzer output.

At first glance, this looks good. The only thing that surprises me is that `STB` changes between `STB` and `ACK` (i.e. in the middle of a transfer). That shouldn't be the case. Maybe this is a bug in the XBUS bridge. I'll have a look at it.

By the way, could we have a look at your SRAM controller? How do you handle the (byte-) address and `STB`? Do you use `STB` only for write-data or also for read-data?

> I suspect this is a bug in driving the wb_sel lines correctly during cpu execution from external memory.

I think so, too... 🙈

stnolting, 2025-05-17:
There is indeed a bug in the `SEL` signal (but this should only occur if the caches are activated). Could you test your setup again with the current version (after mergin #1270)?

kurtjensen, 2025-05-19:
Hello!
I'm not using the packaged Vivado model with AXI.

I had checked all three data widths (aligned, of course) with the SRAM test I cooked up.

I did not yet try writing a linker script that mapped code to IRAM and data to external SRAM.  

The SRAM controller honors the SEL signals for either read or write, 8, 16 or 32 bit transfers.  It samples wb_stb[3:0] once on the clock edge where wb_stb falls, and then sticks with that until wb_ack is asserted by the same code.  I hope that is the right thing to do :)

I had similar woes on a Spartan 7 board too, but never got to using the ILA on that board.

Thanks for having a look.  I'll try that new version when I am back later on this week. Your help is much appreciated!

kurtjensen, 2025-05-21:
I pulled/cloned the core directory from 0ac472fc92f9c629aa56b8cb7877fd3863abe1e2 and rebuilt.
I'm sorry to report that it looks like this bug is still there:  8 bit instruction fetch at clock cycle labeled #1554 below:

[image]

If there is a signal you'd like to see in the ILA trace, please let me know :)
-K

stnolting, 2025-05-22:
> Powers that be are good with posting the controller. It can be found at: https://github.com/kurtjensen/wb_psram

That's great! Thanks for sharing this!

> I pulled/cloned the core directory from https://github.com/stnolting/neorv32/commit/0ac472fc92f9c629aa56b8cb7877fd3863abe1e2 and rebuilt.
I'm sorry to report that it looks like this bug is still there: 8 bit instruction fetch at clock cycle labeled #1554 below:

`wb_sel` is still changing during an actual bus request. I think I've found the cause of this bug (#1272).

The CPU's front-end had `ben` (which is equivalent to Wishbone's `stb`) hardwired to all-zero. In the past, I understood the Wishbone spec to mean that `ben`/`stb` is only relevant for write accesses. With the fix the CPU will always request full 32-bit words (for instruction fetches only). In addition, the internal bus system will now also pass this information on correctly to all other modules, including the Wishbone port.

closed 2025-05-23 by stnolting

kurtjensen, 2025-05-24:
I did find the wishbone spec confusing to read.  When I saw this bit:

> 3.2.2 Classic pipelined SINGLE READ Cycle
> 
> The bus protocol works as follows:
>   CLOCK EDGE 0
> ◦ MASTER presents a valid address on [ADR_O()] and [TGA_O()].
> ◦ MASTER negates [WE_O] to indicate a READ cycle.
> ◦ MASTER presents bank select [SEL_O()] to indicate where it expects data.
> ◦ MASTER asserts [CYC_O] and [TGC_O()] to indicate the start of the cycle.
> ◦ MASTER asserts [STB_O] to indicate the start of the phase.
>   CLOCK EDGE 1
> ◦ SLAVE decodes inputs, and responding SLAVE asserts [ACK_I].
> ◦ SLAVE presents valid data on [DAT_I()] and [TGD_I()].
> ◦ SLAVE asserts [ACK_I] in response to [STB_O] to indicate valid data.
> ◦ MASTER negates [STB_O] to indicate end of data phase.
> ◦ MASTER monitors [ACK_I], and prepares to latch data on [DAT_I()] and [TGD_I()].
>   CLOCK EDGE 2
> ◦ MASTER latches data on [DAT_I()] and [TGD_I()].
> ◦ MASTER negates [CYC_O] to indicate the end of the cycle.
> ◦ SLAVE negates [ACK_I].

I interpreted that as the slave only being allowed present read cycle data on the data lines indicated by sel_o[3:0]. And, that once stb_o is de-asserted then the master was free to change sel_o[3:0].  sel_o[3:0] needs to be latched by the slave when stb_o is still asserted, so the slave controller can then respond appropriately during subsequent wait states before asserting ack_i and ending the cycle.

This easily could have been a misinterpretation on my part :). I'm a wishbone newbie, I suppose.

Thanks again for looking into this!

kurtjensen, 2025-05-27:
I pulled 91a1fe3c28af572af0c7cb79abad36a325cf3111 and found good news: Instruction fetches from the external memory are working now with no 8 bit fetches! Brilliant!  Thank you!  

It can now get all the way to the GPIO LED blinking loop at the end of a hello world app running out of external SRAM.

I'm now seeing a intermittent trap I didn't expect occur occasionally while blinking the LED in that loop, but at least now the RTE code is getting far enough to report it to the serial port quite nicely.  I'll debug further and gather more evidence before choosing to post another bug.

With your comment above about only fetching 32 bits of instructions at a time, I wonder about the use of RV32C 16 bit instructions.
As I understand it (which may be a poor understanding), the 32 bit instructions only need 16 bit alignment to be able to pack them in memory usefully.  It is reasonable to expect that 16 bit aligned 32 bit instructions may need two fetches but should otherwise function properly?

Thanks again.  You and your contributors have done fantastic work here, and I really do appreciate it!

kurtjensen, 2025-05-28:
With regard to the trap mentioned in the previous post. Some hardware debugging revealed a bad IC. With replacement, execution from external wishbone memory seems to be working well, so please disregard that bit.  Cheers!

stnolting, 2025-05-31:
> I pulled https://github.com/stnolting/neorv32/commit/91a1fe3c28af572af0c7cb79abad36a325cf3111 and found good news: Instruction fetches from the external memory are working now with no 8 bit fetches! Brilliant!

Great to hear! :+1:

> I'm now seeing a intermittent trap I didn't expect occur occasionally while blinking the LED in that loop

Which trap / which `mcause` value?

> the 32 bit instructions only need 16 bit alignment to be able to pack them in memory usefully

That is correct.

> It is reasonable to expect that 16 bit aligned 32 bit instructions may need two fetches but should otherwise function properly?

This is the really annoying part of the C ISA extension... 16-bit and 32-bit instructions can be mixed freely. So a 32-bit instruction word can also be 16-bit aligned. But then you need two memory accesses that load the individual 16-bit parts and reassemble them with a special logic (zhis is also the reason why we have an instruction prefetch buffer / FIFO).

> Thanks again. You and your contributors have done fantastic work here, and I really do appreciate it!

Thank you very much for the nice words. :)

kurtjensen, 2025-05-31:
I suppose that was ambiguous. The memory chip itself was somehow flaky. Replacement made operation reliable.Thanks again!

Bad IC = bad instruction counter or bad IC = broken chip? 😅

stnolting, 2025-06-01:
So I think we can close this, right? ;)

kurtjensen, 2025-06-01:
Yes, I think it would be appropriate to close this at this point.  Thanks!