stnolting/neorv32 #1271: [idea] `DMA` bug, `CRC` relevance, `CFS` makeover

stnolting, 2025-05-18:
The DMA probably also has a problem with the `ben` signal (see #1269), which has incorrect values during read accesses. A fix would therefore be necessary.

This leads me to consider whether the **DMA** is necessary at all. The same applies to the **CRC** module. Both modules probably only provide an advantage for very specific tasks. To really use them effectively and in a generic way, the DMA in particular would have to offer significantly more features (multiple channels, burst support, automatic triggers, etc.).

I am thinking about removing DMA and CRC. CRC can be calculated very well with a custom instruction. And you could theoretically also use the second CPU core as a (larger, maybe slower, but much more capable) DMA replacement.

Beyond this, I would find it much more exciting if the custom functions subsystem itself had access to the bus. This would allow you to build a DMA of any complexity, for example. Perhaps it would be even better if the system bus could be "passed through" the CFS. This would still give you the option of building a DMA with it (a simple bus mux built into the system bus), but you would also have additional options such as "bus sniffing". 

These are just ideas so far... What do you think?

Unike267, 2025-05-20:
Hi Stephan!

> Both modules probably only provide an advantage for very specific tasks

As you mentioned, this module could be necessary in specific situations. For example, when you have an application working continuously and needing to dump data to other memory-mapped peripherals in the background constantly.

Alternatively, you could use external memory-mapped DMA modules, like the [LiTeX DMA module](https://github.com/enjoy-digital/litex/blob/master/litex/soc/cores/dma.py), but you'd need to ensure it's compatible with your specific application.  For example, with the current built-in DMA, you can access any memory region for transfers, including DMEM and IMEM, right?  We should consider whether an external DMA would offer the same flexibility and access to these regions.

> I would find it much more exciting if the custom functions subsystem itself had access to the bus. 

Wow! This is a very interesting feature!

> Perhaps it would be even better if the system bus could be "passed through" the CFS.

One consideration that comes to mind is the need for some kind of semaphore or arbitration mechanism to manage bus access.  If the CPU is very busy, the execution of the CFS might get significantly delayed. At the moment, CFS is essentially a “doped” memory-mapped connection, so in practice, and in my opinion, there's little difference between attaching an accelerator through it or directly to the XBUS. Maybe it would bring a fresh 🧊 perspective to the project to rethink the architecture of this connection mode a bit.

Cheers! 😃

stnolting, 2025-05-26:
I use external DMAs a lot. Basically, they do all the work and the CPU just orchestrates them. The only thing is that an external DMA cannot reach the processor-internal modules - and I do not like adding a device port to the processor...

So perhaps the current DMA deserves a second chance. Multiple channels would be good, but they also cost a lot of hardware. Maybe you can let the DMA fetch the descriptors from memory itself. 🤔

However. The CRC module seems to be far too specific. It also works iteratively (yes, it is nice and small as a result), but it also blocks the bus during this time, which reduces the overall performance. So I think it's time for the CRC module to leave the party.

> One consideration that comes to mind is the need for some kind of semaphore or arbitration mechanism to manage bus access.

We do have a primitive for that. The bus multiplexer features two option: always prioritize a port (i.e. the CPU) or use a "fair" round-robin approach where the accesses alternate.

https://github.com/stnolting/neorv32/blob/e5ab2e845fe6b030348bc2589161dc208a1f4177/rtl/core/neorv32_bus.vhd#L17-L22

Unike267, 2025-05-27:
> So perhaps the current DMA deserves a second chance. 

Great! 

> However. The CRC module 

Okay, that makes perfect sense!

> We do have a primitive for that. The bus multiplexer features two option: always prioritize a port (i.e. the CPU) or use a "fair" round-robin approach where the accesses alternate.

Wow! I didn't know about this feature. Thanks for the info! (I love learning new things about the micro 😃)