stnolting/neorv32 #1311: bootloader fails to build with stack alignment error

htminuslab, 2025-07-10:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment:**
 - OS
 - GCC Version (RISC-V and native)
 - Libraries used (clib, newlib, ...)

**Hardware:**
 - Hardware version (x.x.x.x)
 - Implemented CPU extensions, peripherals, ...
 - Hardware modifications

**Additional context**
Add any other context about the problem here.

Unike267, 2025-07-11:
Hi there!

Could you follow these steps and tell me if the result obtained is satisfactory?

# To compile through sim container

- apt install podman
- podman pull ghcr.io/stnolting/neorv32/sim:latest
- git clone https://github.com/stnolting/neorv32
- cd neorv32
- podman run --rm -itv $(pwd):/wrk:Z -w /wrk stnolting/neorv32/sim:latest make -C sw/example/hello_world/ exe

# To run bootloader in arty a7 35t (I don't know the FPGA you have)

- apt install podman openFPGALoader
- podman pull ghcr.io/stnolting/neorv32/sim:latest
- git clone --recursive https://github.com/stnolting/neorv32-setups
- cd neorv32-setups/neorv32
- podman run --rm -itv $(pwd):/wrk:Z -w /wrk stnolting/neorv32/sim:latest make -C sw/example/hello_world/ exe
- cd ../vivado/arty-a7-test-setup
- source /*your_vivado_installation_absolute_path*/Vivado/*20XX.X*/settings64.sh
- vivado -mode batch -nojournal -nolog -source create_project.tcl
- openFPGALoader --board arty work/arty-a7-35-test-setup.runs/impl_1/neorv32_test_setup_bootloader.bit 

Load `exe` through a graphical serial terminal.

> [!IMPORTANT]
> I assumed that you have Vivado installed. 

Cheers!

htminuslab, 2025-07-11:
Hi stnolting/Unike267,

Thanks for trying help, here is the sequence I use:

```
hans@Linux586:~$ git clone https://github.com/riscv/riscv-gnu-toolchain
hans@Linux586:~$ cd riscv-gnu-toolchain
hans@Linux586:~$ riscv-gnu-toolchain$ ./configure --prefix=/opt/riscv --with-arch=rv32i --with-abi=ilp32
hans@Linux586:~$ riscv-gnu-toolchain$ make
hans@Linux586:~$ riscv32-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv32-unknown-elf/15.1.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /home/hans/neorv32/riscv-gnu-toolchain/gcc/configure --target=riscv32-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=ilp32 --with-arch=rv32i --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 15.1.0 ()
hans@Linux586:~$ git clone https://github.com/stnolting/neorv32.git
cd neorv32/sw/example/demo_blink_led/
```
next I added "RISCV_PREFIX ?= riscv32-unknown-elf-" to the makefiles.
```
hans@Linux586:~/neorv32/neorv32/sw/example/demo_blink_led$ make check
---------------- riscv32-unknown-elf-gcc ----------------
Using built-in specs.
COLLECT_GCC=riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv32-unknown-elf/15.1.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /home/hans/neorv32/riscv-gnu-toolchain/gcc/configure --target=riscv32-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion= --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././gcc --disable-multilib --with-abi=ilp32 --with-arch=rv32i --with-tune=rocket --with-isa-spec=20191213 'CFLAGS_FOR_TARGET=-Os    -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os    -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib zstd
gcc version 15.1.0 ()
---------------- riscv32-unknown-elf-objdump ----------------
GNU objdump (GNU Binutils) 2.44
---------------- riscv32-unknown-elf-objcopy ----------------
GNU objcopy (GNU Binutils) 2.44
---------------- riscv32-unknown-elf-readelf ----------------
GNU readelf (GNU Binutils) 2.44
---------------- riscv32-unknown-elf-size ----------------
GNU size (GNU Binutils) 2.44
---------------- NEORV32 image_gen ----------------
NEORV32 executable image generator
Three arguments are required.
1st: Operation
 -app_bin : Generate application executable binary (binary file, little-endian, with header)
 -app_vhd : Generate application raw executable memory image (vhdl package body file, no header)
 -bld_vhd : Generate bootloader raw executable memory image (vhdl package body file, no header)
 -raw_hex : Generate application raw executable (ASCII hex file, no header)
 -raw_bin : Generate application raw executable (binary file, no header)
 -raw_coe : Generate application raw executable (COE file, no header)
 -raw_mem : Generate application raw executable (MEM file, no header)
 -raw_mif : Generate application raw executable (MIF file, no header)
2nd: Input file (raw binary image)
3rd: Output file
4th: Project name or folder (optional)
---------------- Native GCC ----------------
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/11/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 11.4.0-1ubuntu1~22.04' --with-bugurl=file:///usr/share/doc/gcc-11/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-11 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-11-XeT9lY/gcc-11-11.4.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-11-XeT9lY/gcc-11-11.4.0/debian/tmp-gcn/usr --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-serialization=2
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.4.0 (Ubuntu 11.4.0-1ubuntu1~22.04)

Toolchain check OK
hans@Linux586:~/neorv32/neorv32/sw/example/demo_blink_led$ make bin
[... cut ...]
Unfortunately the bootloader still fails:

```
hans@Linux586:~/neorv32/neorv32/sw/bootloader$ make bootloader
/opt/riscv/lib/gcc/riscv32-unknown-elf/15.1.0/../../../../riscv32-unknown-elf/bin/ld: error: /opt/riscv/lib/gcc/riscv32-unknown-elf/15.1.0/../../../../riscv32-unknown-elf/lib/libc.a(libc_a-ctype_.o) use 16-byte stack aligned but the output use 4-byte stack aligned
```

Hi Unike267, 

Thanks for the sequence but my FPGA (Efinix T20Q100) does not have a jtag port, just a UART.

Regards,
Hans.

Unike267, 2025-07-11:
>  does not have a jtag port, just a UART.

Nothing is mentioned about jtag, the connection is through UART.

Please try to build bootloader through sim container, follow these steps:

- podman pull ghcr.io/stnolting/neorv32/sim:latest
- git clone https://github.com/stnolting/neorv32
- cd neorv32
- `sed -i '244s|    "Available CMDs:|    "TEST-able CMDs:|' sw/bootloader/main.c`
- podman run --rm -itv $(pwd):/wrk:Z -w /wrk stnolting/neorv32/sim:latest make -C sw/bootloader/ bootloader
- synthesize your design with the Efinix tool and load it in your FPGA, verify with the execution of `hello_word` program.

Everything works normally for me:

```
NEORV32 Bootloader

BLDV: Jul 11 2025
HWV:  0x01110801
CLK:  0x05f5e100
MISA: 0x40801104
XISA: 0x00000083
SOC:  0x0003800d
IMEM: 0x00004000
DMEM: 0x00002000

Auto-boot in 10s. Press any key to abort.
Aborted.

TEST-able CMDs:
 h: Help
 r: Restart
 u: Upload via UART
 s: Store to SPI flash
 l: Load from SPI flash
 e: Start executable
CMD:> u
Awaiting neorv32_exe.bin... OK
CMD:> e
Booting from 0x00000000...

                                                                                      ##        ##   ##   ##    
 ##     ##   #########   ########    ########   ##      ##   ########    ########     ##      ################  
####    ##  ##          ##      ##  ##      ##  ##      ##  ##      ##  ##      ##    ##    ####            ####
## ##   ##  ##          ##      ##  ##      ##  ##      ##          ##         ##     ##      ##   ######   ##  
##  ##  ##  #########   ##      ##  #########   ##      ##      #####        ##       ##    ####   ######   ####
##   ## ##  ##          ##      ##  ##     ##    ##    ##           ##     ##         ##      ##   ######   ##  
##    ####  ##          ##      ##  ##      ##    ##  ##    ##      ##   ##           ##    ####            ####
##     ##    #########   ########   ##       ##     ##       ########   ##########    ##      ################  
                                                                                      ##        ##   ##   ##    
Hello world! :)
```

htminuslab, 2025-07-11:
Hi Unai,

Again thanks for your help, but I might have not been clear, there is no issue with loading the design and I have it up and running on my board with zero effort (kudos to snolting). However, it defaults to 19200 which is not an easy baudrate for me and hence I would like to re-compile the bootloader. 

I do not believe I have to go though a container to do this and I have used the bare-metal setup for my own core without any issues. Also all the examples I tried worked (e.g. coremark) but not the bootloader.  I will do some digging over the weekend and will report back If I find the solution,

Regards,
Hans.

stnolting, 2025-07-11:
@htminuslab 

> Also all the examples I tried worked (e.g. coremark) but not the bootloader. I will do some digging over the weekend and will report back If I find the solution,

This is very strange... 🤔

A long time ago I also had problems with this stack alignment, but I can't really remember what the reason was or how I solved it.

> So I rebuild gcc using stnolting suggestion:

I haven't built a toolchain myself for a long time. Maybe something has changed in the build process?! However, your `-v` output actually looks fine: ABI is ok, ARCH is okand the toolchain uses newlib.

Could you try out the pre-built toolchain from xPack?
https://github.com/xpack-dev-tools/riscv-none-elf-gcc-xpack/releases/tag/v14.2.0-3

htminuslab, 2025-07-12:
Hi All,

I found an easy solution, simply not use the minimum register version:

```
#MARCH = rv32e_zicsr_zifencei
#MABI = ilp32e

MARCH = rv32i_zicsr_zifencei
MABI = ilp32

hans@Linux586:~/neorv32/neorv32/sw/bootloader$ make bootloader
Memory utilization:
   text    data     bss     dec     hex filename
   2508       0       4    2512     9d0 main.elf
Generating neorv32_bootloader_image.vhd
Installing bootloader image to ../../rtl/core/neorv32_bootloader_image.vhd
```

I am not sure why my gcc switches to 4-byte stack alignment with rv32e but I guess most of us don't need it.  

The bootloader now builds and runs fine:

```
# ** Note: [NEORV32] Implementing processor-internal IMEM as blank RAM.
#    Time: 0 ps  Iteration: 0  Instance: /neorv32_processortop_minimalboot_tb/U_0/neorv32_inst/memory_system/neorv32_int_imem_enabled/neorv32_int_imem_inst
# ** Warning: [NEORV32] Assuming this is a simulation.
#    Time: 0 ps  Iteration: 0  Region: /neorv32_processortop_minimalboot_tb/U_0/neorv32_inst/core_complex_gen(0)/neorv32_cpu_inst/hello_neorv32
# ** Note: [NEORV32] CPU tuning options: 
#    Time: 0 ps  Iteration: 0  Region: /neorv32_processortop_minimalboot_tb/U_0/neorv32_inst/core_complex_gen(0)/neorv32_cpu_inst/hello_neorv32
# ** Note: [NEORV32] CPU ISA: rv32ix_zicntr_zicsr_zifencei
#    Time: 0 ps  Iteration: 0  Region: /neorv32_processortop_minimalboot_tb/U_0/neorv32_inst/core_complex_gen(0)/neorv32_cpu_inst/hello_neorv32
# ** Note: [NEORV32] BOOT_MODE_SELECT = 0: booting via bootloader
#    Time: 0 ps  Iteration: 0  Region: /neorv32_processortop_minimalboot_tb/U_0/neorv32_inst/sanity_checks
# ** Note: [NEORV32] Processor Configuration: CPU (single-core) IMEM DMEM BOOTROM CLINT GPIO UART0 PWM SYSINFO 
#    Time: 0 ps  Iteration: 0  Region: /neorv32_processortop_minimalboot_tb/U_0/neorv32_inst/sanity_checks
# ** Note: [NEORV32] The NEORV32 RISC-V Processor (v01.11.07.09), github.com/stnolting/neorv32
#    Time: 0 ps  Iteration: 0  Region: /neorv32_processortop_minimalboot_tb/U_0/neorv32_inst/sanity_checks
# N =  1, M = 128, O = 2, DIV0 =  16, DIV1 =   2, DIV2 =   2
# Using internal feedback
# CLKIN reference clock period = 83.3340 ns (12.00 MHz)
# VCO period = 0.6510 ns (1535.99 MHz)
# PLL period = 1.3021 ns (767.99 MHz)
# CLKOUT0 period = 20.8335 ns (48.00 MHz), shift = 0.0000 ns
# CLKOUT1 period = 2.6042 ns (384.00 MHz), shift = 0.0000 ns
# CLKOUT2 period = 2.6042 ns (384.00 MHz), shift = 0.0000 ns
# Reached Lock @ 1041.67 ns
# UART0: 
# UART0: NEORV32 Bootloader
# UART0: 
# UART0: BLDV: Jul 12 2025
# UART0: HWV:  0x01110709
# UART0: CLK:  0x02dc6c00
# UART0: MISA: 0x40800100
# UART0: XISA: 0x80000083
# UART0: SOC:  0x0013800d
# UART0: IMEM: 0x00010000
# UART0: DMEM: 0x00010000
# UART0: 
# UART0: Available CMDs:
# UART0:  h: Help
# UART0:  r: Restart
vsim 16>
```

Thanks,
Hans.

closed 2025-07-12 by htminuslab

stnolting, 2025-07-14:
Nice to hear that it works!
But it is strange... I use the toolchain from xPack and there are no problems with the `E` ISA extensions or the according ABI / stack alignment. 🤔