stnolting/neorv32 #1323: Prioritizing S_RESTART to S_REQUEST in frontend

mmcheraghi, 2025-07-24:
**Describe the bug**
In case of free space in ipb and incoming if_reset, S_REQUEST state goes to S_PENDING state in case of S_RESTART. 

        when S_REQUEST => -- request next 32-bit-aligned instruction word
        -- ------------------------------------------------------------
          fetch.restart <= fetch.restart or ctrl_i.if_reset; -- buffer restart request
          if (ipb.free = "11") then -- free IPB space?
            fetch.state <= S_PENDING;
          elsif (fetch.restart = '1') or (ctrl_i.if_reset = '1') then -- restart because of branch
            fetch.state <= S_RESTART;
          end if;
https://github.com/stnolting/neorv32/blob/b1011b0511ae1221fa93f966c7606954a34058ac/rtl/core/neorv32_cpu_frontend.vhd#L85C1-L92C18

**Expected behavior**
The suggestion is to prioritize S_RESTART state like this:

        when S_REQUEST => -- request next 32-bit-aligned instruction word
        -- ------------------------------------------------------------
          fetch.restart <= fetch.restart or ctrl_i.if_reset; -- buffer restart request
          if (fetch.restart = '1') or (ctrl_i.if_reset = '1') then -- restart because of branch
            fetch.state <= S_RESTART;
          elsif (ipb.free = "11") then -- free IPB space?
            fetch.state <= S_PENDING;
          end if;

stnolting, 2025-07-25:
> We can also change the following lines

Unfortunately, we have the same problem here. If we go directly to `S_RESTART` and do not wait for the end of the currently pending bus access (via `ibus_rsp_i.ack`), we could receive the wrong response during the next fetch.

Anyway, thanks for the ideas! I'm happy someone is diving so deep into the pipeline and trying to improve things! 😃

mmcheraghi, 2025-07-25:
> Is this more of a suggestion to improve performance, or is it really a bug?

You're right, it's a performance optimization suggestion.

Ok, only one more question. Does it gain benefit if we modify the mentioned bus request in the 
following manner and then prioritize switching to S_RESTART in S_REQUEST state?

`ibus_req_o.stb   <= '1' when (fetch.state = S_REQUEST) and (ipb.free = "11") and (ctrl_i.if_reset = '0') else '0';`

Thanks

mmcheraghi, 2025-07-25:
I think the solution is to divide fetch state machine into comb. and sync. parts like the case in cpu_control module. Then we can use fetch_nxt.state = EX_PENDING for bus_stb.

`ibus_req_o.stb   <= '1' when (fetch_nxt.state = S_PENDING) else '0';`

```
  -- Fetch Engine FSM -----------------------------------------------------------------------
  -- -------------------------------------------------------------------------------------------
  fetch_fsm_sync: process(rstn_i, clk_i)
  begin
    if (rstn_i = '0') then
      fetch.state   <= S_RESTART;
      fetch.restart <= '1'; -- reset IPB and issue engine
      fetch.pc      <= (others => '0');
      fetch.priv    <= priv_mode_m_c;
    elsif rising_edge(clk_i) then
      fetch <= fetch_nxt;
    end if;

  end process fetch_fsm_sync;

  fetch_fsm_comb: process(fetch, ctrl_i, ipb, ibus_rsp_i)
  begin
    fetch_nxt <= fetch;
    case fetch.state is

      when S_REQUEST => -- request next 32-bit-aligned instruction word
      -- ------------------------------------------------------------
        fetch_nxt.restart <= fetch.restart or ctrl_i.if_reset; -- buffer restart request
        if (fetch.restart = '1') or (ctrl_i.if_reset = '1') then -- restart because of branch
          fetch_nxt.state <= S_RESTART;
        elsif (ipb.free = "11") then -- free IPB space?
          fetch_nxt.state <= S_PENDING;
        end if;

        when S_PENDING => -- wait for bus response and write instruction data to prefetch buffer
        -- ------------------------------------------------------------
          fetch_nxt.restart <= fetch.restart or ctrl_i.if_reset; -- buffer restart request
          if (ibus_rsp_i.ack = '1') then -- wait for bus response
            fetch_nxt.pc    <= std_ulogic_vector(unsigned(fetch.pc) + 4); -- next word
            fetch_nxt.pc(1) <= '0'; -- (re-)align to 32-bit
            if (fetch.restart = '1') or (ctrl_i.if_reset = '1') then -- restart request due to branch
              fetch_nxt.state <= S_RESTART;
            else -- request next linear instruction word
              fetch_nxt.state <= S_REQUEST;
            end if;
          end if;

      when others => -- S_RESTART: set new start address
      -- ------------------------------------------------------------
        fetch_nxt.restart <= '0'; -- restart done
        fetch_nxt.pc      <= ctrl_i.pc_nxt; -- initialize from PC
        fetch_nxt.priv    <= ctrl_i.cpu_priv; -- set new privilege level
        fetch_nxt.state   <= S_REQUEST;

    end case;
  end process fetch_fsm_comb;
```

stnolting, 2025-07-26:
Well spotted! But in this case `fetch_nxt.state` is generated from `ctrl_i.if_reset` via a combinational circuit. The result would therefore be the same as above. :(

closed 2025-07-28 by mmcheraghi