stnolting/neorv32 #1327: [User Guide] Section "1. Software Toolchain Setup", incorrect commands

CyberFox001, 2025-07-30:
In the User Guide, section "1. Software Toolchain Setup", there is a list of commands under subsection "Building the Toolchain from Scratch".

The `git clone' command use a different URL than the one described in the text, before the commands.
In the text, the URL is "https://github.com/riscv-collab/riscv-gnu-toolchain" and in the `git clone'  command the URL is "https://github.com/riscv/riscv-gnu-toolchain".

Additionally, if I configure the toolchain build with the option given in the User Guide:
```
./configure --prefix=/opt/riscv --with-arch=rv32i --with-abi=ilp32
``` 

After the `make' command I obtain binaries that have the prefix "riscv32-unknown-elf-" instead of "riscv-none-elf-", as tell in the User Guide and also set by default in the file sw/common/common.mk.

CyberFox001, 2025-07-30:
I built the blink led example like described in [section "3. Application Program Compilation" of the User Guide](https://stnolting.github.io/neorv32/ug/#_application_program_compilation). I only modified the `RISCV_PREFIX' variable to match the new toolchain binaries prefix:
```
make RISCV_PREFIX="riscv32-unknown-elf-" clean_all exe
```

I can build, but when I upload to the bootloader (with cutecom) I get an error from the bootloader:
```
Awaiting neorv32_exe.bin... ERROR_SIGNATURE
```

stnolting, 2025-07-30:
Hmm, this is strange. Could you post the executable (`neorv32_exe.bin`) you are trying to upload here (`neorv32_exe.bin`)? Then I can check the file format.

CyberFox001, 2025-07-30:
Here a copy of the `neorv32_exe.bin` built from the example "demo_gpio": https://github.com/CyberFox001/things/blob/main/neorv32_exe.bin

stnolting, 2025-07-30:
The signature looks fine. The error must be caused by the terminal program. 🤔 
Could you try another program (or perhaps the upload script)?

By the way, are you sure that's the demo_gpio program? In the source code, I saw something about "dual core demo":

[image]

stnolting, 2025-08-01:
> it worked with the upload script. So, I will use it in the future.

That's great! But minicom _should_ work... 🤔
I'll test that again.

stnolting, 2025-09-06:
`minicom` always seems to use a protocol, even if you select “ascii”...

I experimented with `piococom` and that seems to work:

```
$ picocom --b 19200 /dev/ttyUSB1 --send-cmd "cat"

picocom v3.1

port is        : /dev/ttyUSB1
flowcontrol    : none
baudrate is    : 19200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
hangup is      : no
nolock is      : no
send_cmd is    : cat
receive_cmd is : rz -vv -E
imap is        :
omap is        :
emap is        : crcrlf,delbs,
logfile is     : none
initstring     : none
exit_after is  : not set
exit is        : no

Type [C-a] [C-h] to see available commands
Terminal ready

NEORV32 Bootloader
build: Sep  3 2025

Auto-boot in 10s. Press any key to abort.
Aborted.

Type 'h' for help.
CMD:> 
```

Type `u` and press `ctrl-A` `ctrl-S` to open the send-file dialog. Enter the absolute file path and hit enter.

```
CMD:> u
Awaiting neorv32_exe.bin...
*** file: /mnt/n/projects/neorv32/sw/example/demo_gptmr/neorv32_exe.bin
$ cat /mnt/n/projects/neorv32/sw/example/demo_gptmr/neorv32_exe.bin

*** exit status: 0 ***
OK
CMD:> e
Booting from 0x00000000...
```