stnolting/neorv32 #1336: Error in item 21 of processor check

mmcheraghi, 2025-08-13:
Hi
In the latest edition of the processor, item 21 of processor (CLINT.MTI) check fails.

stnolting, 2025-08-13:
Oh what?

The CI did not report any failing checks. In the latest run the CLINT.MTI check seems to work fine:

https://github.com/stnolting/neorv32/actions/runs/16927965097/job/47967384100#step:7:151

mmcheraghi, 2025-08-13:
The problem was with old processor check program that was not updated. Thanks

stnolting, 2025-08-14:
Ah right. I had to adjust the processor checks because of #1333:

> `mtinst` CSR: Do not write zero to the CSR in case of an interrupt. mtinst now also shows the last executed instruction word when encountering an interrupt.

In the latest version the CLINT.MTI test does not longer check if `mtval` is zero.